surberj/Vision #4: Meetings

surberj, 2015-10-13:
20150921 Margarita, Lucas, me
- Q&A for OKVIS and PTAM (see figure)
- main parts of thesis
 - Literature Review on Visual-Inertial SLAM
 - Interface VI_Sensor with code-base of OKVIS, PTAM and potentially others
- General Questions discussed about SLAM
 - Bundle Adjustment
 - Filtering vs Batch Approches
 - Marginalization of Frames, Features, Keyframes
- borrowed book () of Mrgarita

surberj, 2015-10-13:
20151001 Margarita, Lucas, me
- Installing ROS:
 - failed on mac
 - installed on VirtualBox, working but very slow
 - started with Tutorials
 - status: try to build PTAM, error with PtamParamsConfig.h
- Literature:
 - ROVIO, filtering approach, use for comparison with OKVIS
 - Performance measures (out of ROVIO, PTAM and OKVIS): accuracy, robustness, computational complexity, limitations
- Goal of thesis: Compare OKVIS and ROVIO wrt these performance measures
- Outline: 
 - Install Linux on mac-partition
 - PTAM with mac-webcam
 - ROVIO with VI-Sensor
 - OKVIS with VI-Sensor
 - ev other hardware
- For technical questions ping Lucas

surberj, 2015-10-13:
20151006 Margarita, Lucas, me
- Ubuntu 14.04 installed on mac-partition
- ROS working better this way
- Installed PTAM
 - problems with Image processing (auto-focus, color, ...)
 - problems with camera-calibration
- Short intro into PTAM by Lucas
- ros inspection tool: rqt
- Discussion about parameters: static vs variable params
- Decided Lucas will buy a webcam for me

surberj, 2015-10-13:
20151012 Bharat, me
- https://github.com/wingtra/vision_thesis
- dataset of VI-Sensor: https://bitbucket.org/adrlab/rcars/downloads
- Bharat is actually playing with LSD-SLAM, PTAM and RCARS (on SITL and with VI-Sensor dataset)
- Issues with Cameracalibration (distortion models differ for different algorithms)
- I will attend meeting on Thursday with Margarita
- Outlook Wingtra-side
 - Do evaluation of different approaches on Laptops
 - Hardware (Camera, Vision Computer) not fixed yet
  - if available by than work with snapdragon, if not start with IMX-6
  - different cameras are more or less interfaced with IMX-6 by now
- My semester thesis will take place on my laptop (nothing embedded yet)
- My main contribution: useful comparison of different approaches (eventually include LSD-SLAM or RCARS or others as well)
- Eventually my secondary contribution: test algorithms on not time-synchronized hardware

surberj, 2015-10-13:
20151012 Margarita, Lucas, me
- brief report:
 - webcam calibration not perfect but reasonable to continue with PTAM
 - creates this github repo
 - thesis proposal in readme
 - dived into PTAM code - main functionality is clear, explicit calculations not revisited, started short analysis of influence of ptam-parameters
 - meeting with Bharat (mainly dataset from RCARS)
- code of ROVIO: (add link)
- next steps:
 - define performance measures
 - run ROVIO with dataset and secondly with VI-Sensor
 - play around with PTAM (understand limitations, influence of params)

surberj, 2015-10-20:
20151020 Margarita, Lucas, me

- brief report:
 - rovio with bagfiles
 - rovio with VI-Sensor, setup of VI-sensor is working
 - worked on logging/postprocessing (log .bag, convert to .csv, import in matlab)
 - issue with alignment of ROVIO data and VICON data
 - Wingtra: Today we flew successfully with hover position control
- discussed different ways to diverge once more:
 - Investigate in OCVIS, discuss with Lucas
 - Investigate in multisensorfusion of ASL
 - Investigate in graph-SLAM (Eade Drummond)
 - reading SVO, RCARS (already partly done), LSD_SLAM, others?
 - try rovio on different hardware (especially Wingtra related)
- decided not to diverge to much, focus on
 - ROVIO, limits, dependence on params, 
 - OCVIS, try to make it run, investigate also in calibration, play with it, talk to Lucas
- for next week: prepare time plan for thesis (official due date: 22.11.2015): Reading, Experiments with ROVIO, with OCVIS, Writing, Presentation preparation, ...

surberj, 2015-10-27:
20151027 Margarita, Lucas, me
- Agreed on Time schedule (see https://github.com/surberj/Vision/blob/postprocessing/timeplan.pdf)
 - OKVIS will be skipped and replaced by ORB_SLAM (and eventually the loosely coupled approach of Weiss, 2012)
- OKVIS: 
 - state at the moment: running super slow --> not able for pose estimation (same for Lucas than for me)
 - we skip it from my thesis
- ROVIO:
 - Plot 1 for Thesis: absolute trajectory error (mean) and complexity_measure (average execution time per frame) over number of features
 - Limits observed: 
  - Divergence for < 5 features
  - Divergence for > 40 features (for live operation - not slowed down)
  - sensitive to camera calibration
  - Drifting over time (error increasing over time)
  - no loop closure 
 - discussed new tests with ROVIO:
  - include IMU timestamp noise
  - include noise on IMU mesurements
  - rewrite code to run "slower than live" to extend plot 1 to higher number of features
 - meet Michael Blösch to get inputs
- datasets:
 - continue atm mainly with euroc dataset
 - KITTI dataset
- Todo till next week (Milestone 1)
 - install ORB_SLAM and loosely coupled approach of Weiss
 - prepare ROVIO to perform new tests (see above)
 - define desired plots for thesis
 - meet Michael Blösch
 - ROVIO code

surberj, 2015-10-28:
20151028 Michael Blösch, me
- Goal: 
 - discuss evaluations of ROVIO
 - main challenges of ROVIO
 - postprocessing he used for paper reusable?
- Evaluations proposed by me:
 - limits in number of features, performance /complexity depending on number of features
 - drifting issue --> relative pose error / absolut trajectory error depending on travelled distance
 - introduce artificial error in IMU-Timestamp (bias or noise)
 - introduce artificial noise in IMU-Measurement
- additional Evaluations according to his input
 - Height test (limit of how far features can be away)
 - multicamera analysis
 - analyse error characteristics in Vision (intensity noise atm set to 2/256 greyscales)
 - analyse influence of features
- Future of ROVIO
 - GPS and other sensors can be included in ROVIO in the update step
 - Global Map: Rovio sends data to Backend that saves a global map and performes a global optimization (e.g. ASL multi agent mapping - closed code)
- he gives me access to postprocessing repo he used for his paper

devbharat, 2015-10-28:
When do you meet him ? I'd like to join.

surberj, 2015-10-28:
I met him this morning at 10. I am sorry for the bad communication - we discussed the notes above. If you like I can tell you tomorrow in office more about the outcome.

surberj, 2015-10-30:
20151030 Michael Blösch, Raghav, Bharat, Max, Julian

Goal: Meeting to see who is working somehow with ROVIO

Protocol:
- Issue with Bharat: Angle initialization: misalignment of IMU and camera frames make ROVIO diverge.
 - Would be interesting to evaluate how much disalignment the algorithm is able to handle (include ev in Semesterthesis of julian)
 - Ros assumes camera frames x-axis aligned with optical axis, normal convention (e.g. convention of simulation) assumes camera z-axis aligned with optical axis --> Raqhav solved the transformations, see picture attached
 - ROVIO estimates the Extrinsics (transformation between camera frame and imu frame) online. Can be fixed if well known (in simulation it is perfectly known, out of kalibr you get also for real world reasonable values). Ideas on that:
  - Especially when ROVIO is started when you have little motion this could be valuable
   - Fix only the transitional part, estimate the attitude
   - Start ROVIO only after take-off
   - Fix the transformation as long as you do not detect motion, start estimation as soon as you have motion
- Bharat: Depth estimation is based on triangulation, could be bad for our task of only moving along the optical axis
 - Michael: few centimeters are enough to convert, probably not a big issue
 - Raghav: Depth estimation is in simulation quite good
 - Evaluation could be useful
- Tag landing based on simple tag recognition and fusing them into msf framework works reasonable
 - workspace 5cm - 5m
- Include Tags in ROVIO?
 - Michael: Attention with loop closures --> rerecognized tags could lead to jumps in position estimate
 - Alternative way: use back-end (like multi_agent_mapping of ASL, closed source). Work with two inertial frames (one "odometry frame", one "world frame"), use odometry frame for low level control, world frame for high level control, estimate taro between the two.

[image]
Remark to the picture:
- q_ob in "roll-pitch-yaw"

surberj, 2015-11-02:
20151102 Margarita, Lucas, me

- discussed different "Drafts"
- upcoming week: focus on
 - ROVIO, set up framework: Set inputs (params, calibration, dataset, TEST-ID) --> launch ROVIO --> log Data --> run python-postprocessing --> generate Plots --> save data
 - perform tests 1 and 2 on euroc dataset
- meeting next week:
 - show framework
 - show first plots
 - discuss datasets
 - discuss next evaluations to be done

surberj, 2016-01-06:
The last few meetings for the semester thesis have not been reported here. The presentation took place at 2015/12/07 (Margarita, Lucas, me), a last meeting for discussions about the report on 2015/12/16 (Margarita, me).

closed 2016-01-06 by surberj