surberj/Vision #5: catkin_make error

surberj, 2015-10-14:
The catkin_make command creates an error:
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolatio

while both
catkin build -DCMAKE_BUILD_TYPE=RELEASE 
and
catkin_make_isolated
are working.

My src folder:
julian@julian-UbuntuPartition:~/catkin_ws/src$ ls -l
total 28
drwxrwxr-x 4 julian julian 4096 Okt  4 13:29 beginner_tutorials
lrwxrwxrwx 1 julian julian   49 Okt 13 14:02 CMakeLists.txt -> /opt/ros/indigo/share/catkin/cmake/toplevel.cmake
drwxrwxr-x 7 julian julian 4096 Okt  4 14:00 ethzasl_ptam
drwxrwxr-x 7 julian julian 4096 Okt 14 12:01 kindr
drwxrwxr-x 5 julian julian 4096 Okt 14 11:55 lightweight_filtering
drwxrwxr-x 5 julian julian 4096 Sep 30 10:40 rcars
drwxrwxr-x 9 julian julian 4096 Okt 14 11:47 rovio
drwxrwxr-x 7 julian julian 4096 Okt  7 14:43 usb_cam

The problem was introduced with adding kindr to the workspace

surberj, 2015-10-14:
According to https://github.com/ethz-asl/kindr you do not seem to be able to use catkin_make with kindr, seems to be a problem upstream

weblucas, 2015-10-14:
compile the kindr outside of the catkin workspace and install in your system. use the section (Building with cmake) in the kindr readme

devbharat, 2015-10-14:
@surberj why wouldn't you use catkin_make_isolated ?

surberj, 2015-10-14:
I am currently working with either  catkin build -DCMAKE_BUILD_TYPE=RELEASE or  catkin_make_isolated, both working fine for me. Thanks @weblucas for the hint how to fix it, will do so eventually later.

closed 2015-10-14 by surberj