surberj/Vision #6: Matlab Post Processing

surberj, 2015-10-18:
General way for post processing in matlab:
- log ros session to .bag file
- convert .bag file to several .csv files (see )
- load .csv files with a main script
- do analyses in matlab (position and angle tracking, ...)

Write a routine to read .csv files in matlab.

surberj, 2015-10-18:
PR: Postprocessing #9

devbharat, 2015-10-18:
You can also try to load bag files directly in MATLAB if you have the robotics toolbox. See http://ch.mathworks.com/help/robotics/examples/working-with-rosbag-logfiles.html

surberj, 2015-10-20:
[image]

surberj, 2015-11-03:
Decided to go on with python / c++ for analyses, see #14

closed 2015-11-03 by surberj