surberj/Vision #10: ROVIO performance

surberj, 2015-10-20:


surberj, 2015-10-20:
Raw logged position and orientation of Rovio vs Vicon for cars table dataset:
[image]

surberj, 2015-10-22:
Main issues with ROVIO: drift and no loop closure detection (as also @devbharat was mentionning)

One idea that comes to my mind to enhance ROVIO performance (without Tag reliance):

Use ROVIO as a "visual odometry" algorithm and an additional 'lightweight' algorithm (see below) to detect loop closures. In a multi sensor fusion module the loop closure detector could be used for a pose update (similar to GPS update).

1) Include ideas of FAB-MAP (http://ijr.sagepub.com/content/27/6/647.short). Use place recognition based in images to detect loop closures.
2) Safe the ROVIO output (camera pose and associated features, with uncertainty) every few meteres or seconds as "descriptors", perform a recognition (classifier, bag-of-words, inverse-file-index), if a match is found estimate pose and perform update in multi sensor fusion module.

surberj, 2015-10-22:
ROVIO does not seem to be able to handle other camera calibrations than 'equidistant'. 
E.g. on euroc dataset they provide their camera calibration with a radial-tangential distortion model. Passing this calibration with 'distortion_model: radial-tangential' makes the estimator diverging, passing the calibration with 'distortion_model: equidistant' makes it performing worse than with lucas visensor-calibration (tracking diverges after one minute or so).

The same thing for the rcars dataset. They provide a calibration with a plumb-bob distortion model. Passing it with 'distortion_model: plumb-bob' makes it diverging immediately, passing it with 'distortion_model: equidistant' leeds to slightly worse (but nearly similar) tracking than lucas calibration.

Investigate in code - should be able to work with other models...

devbharat, 2015-10-22:
As far as I remember plumbbob should work, like radtan. I've been using that. Plumbbob has 5 constants I think.

surberj, 2015-10-27:
euroc, lucas calibration, 25 features:
[image]

compared_pose_pairs 2771 pairs
absolute_translational_error.rmse 0.280998 m
absolute_translational_error.mean 0.258391 m
absolute_translational_error.median 0.240620 m
absolute_translational_error.std 0.110424 m
absolute_translational_error.min 0.049174 m
absolute_translational_error.max 0.511280 m

euroc, lucas calibration, 30 features:
[image]

compared_pose_pairs 2746 pairs
absolute_translational_error.rmse 0.180225 m
absolute_translational_error.mean 0.162777 m
absolute_translational_error.median 0.151309 m
absolute_translational_error.std 0.077361 m
absolute_translational_error.min 0.023452 m
absolute_translational_error.max 0.353734 m

surberj, 2015-10-27:
euroc, lucas calibration, 40 features
[image]

compared_pose_pairs 2603 pairs
absolute_translational_error.rmse 0.271722 m
absolute_translational_error.mean 0.249696 m
absolute_translational_error.median 0.232860 m
absolute_translational_error.std 0.107167 m
absolute_translational_error.min 0.017728 m
absolute_translational_error.max 0.558950 m

CPU load: 60 - 95 %

surberj, 2015-10-27:
###### euroc, lucas calibration, 15 features
[image]

compared_pose_pairs 2784 pairs
absolute_translational_error.rmse 0.253315 m
absolute_translational_error.mean 0.217451 m
absolute_translational_error.median 0.173743 m
absolute_translational_error.std 0.129937 m
absolute_translational_error.min 0.015270 m
absolute_translational_error.max 0.589620 m

CPU load: 26 - 33 %

devbharat, 2015-10-28:
Their is a choice in the ROVIO code to enable/disable the "direct" method (chooses reprojection error instead of intensity error for innovation). It'd be interesting to see how this effects accuracy/computation time. Have you already done that or discussed it previously ?

surberj, 2016-01-06:
ROVIO performance on the IJRR set, the EUROC set and my own sets is reported in https://github.com/surberj/Vision/tree/master/thesis/Report

closed 2016-01-06 by surberj