surberj/Vision #15: ROVIO limit: artificial imu-timestamp bias

surberj, 2015-11-09:
Tracking results, Euroc Dataset, IMU-timestamp bias = 0ms, 70ms, 100ms
[image]

ATE_RMSE for biases from 0 to 100 ms
[image]

devbharat, 2015-11-09:
Do you think if it is even possible to have timestamp bais as filter state with no additional measurements ?

surberj, 2015-11-09:
I was also thinking about that idea... Could be an interesting approach but I doubt that this state would be observable. What do you think?

surberj, 2015-11-09:
I rather think it needs to be estimated during some intelligent calibration. If there is a constant time sync bias at all --> I do not even know if it is constant, drifting or noisily changing...

surberj, 2016-01-06:
Results in https://github.com/surberj/Vision/tree/master/thesis/Report

closed 2016-01-06 by surberj