surberj/Vision #29: OKVIS

surberj, 2016-01-19:
OKVIS should be running reliably on my laptop.

surberj, 2016-02-01:
New Beta Version avialable! 

See, http://www.doc.ic.ac.uk/~sleutene/software/docs/okvis_ros/1.1.0/ . In addition to the ros version a generic cmake version is also available. We are expected to test it. As soon as test feedback is good the open source version will be released.

surberj, 2016-02-16:
The new Beta Version is running perfectly on my laptop now (with the asl machine hall dataset) :-). Next step will be to try to run it on our own datasets.

surberj, 2016-02-18:
OKVIS fails with data of our setup (on all kind of bags - outdoor, indoor, handheld and flying). It is an issue with timestamps --> 

W0218 18:55:04.849689 14007 ThreadedKFVio.cpp:391] Frame is newer than oldest IMU measurement. Dropping it.
W0218 18:55:04.897428 14007 ThreadedKFVio.cpp:391] Frame is newer than oldest IMU measurement. Dropping it.
W0218 18:55:04.961459 13981 Subscriber.cpp:124] Frame delayed at time 1450115909.415078402
E0218 18:55:04.998121 14007 FrameSynchronizer.cpp:104] Dropping frame with id 1
...

surberj, 2016-02-19:
Summerizing the current state:
- OKVIS is running with VI sensor data (EUROC MH data) --> It estimates consistent poses and pointclouds (both nodes, okvis_node and okvis_node_synchronous, are good)
- OKVIS with our data:
  - not deterministic: If run twice with the same settings/commands different results are achieved
  - a lot of tracking losses
  - resulting poses are completely wrong
  - It looks as if the IMU data is not incorporated correctly --> check cam IMU calibration, check quaternion conventions, check potential timestamp issues
  - dropping first 1-2 frames if run with okvis_node // no printout about frame drop if run with okvis_node_synchronous

surberj, 2016-05-04:
OKVIS is running reliably now with data from skybotix VI-sensor and data from wingtra VI-sensor. With histogram equalization results look clearly better for some parts of the datasets.

closed 2016-05-04 by surberj

maym86, 2016-07-01:
How did you solve the "Frame is newer than oldest IMU measurement. Dropping it." Error?

surberj, 2016-07-06:
@maym86 In the end we solved it on the hardware-side with a time-synchronized setup. The board with the IMU triggers the camera capture based on its timestamp and assign the triggering time + half exposure time to the image. We do it with an IMU rate around 200 Hz and a framerate around 20 Hz, which works reliably.