surberj/Vision #44: Distortion: Most SFM pipelines / bundle adjustment solvers expect a simplified intrinsics model (f, k1, k2)

surberj, 2016-02-25:
This means that the principle point is assumed to lie in the center (cx = cy = 0), the focal length is identical in both directions (fx = fy = f) and a simple radial distortion with two params is assumed (x_ = (1+k1*r^2+k2*r^4)*x). With this simplifications we introduce errors in the 3d reconstruction (no idea how large they are...).

To get around the issue, I could preprocess and undistort the images. If I do that with my own script using the cv::undistort function I get clear quality losses:
original image:

undistorted image:

devbharat, 2016-02-25:
What do you mean quality losses ?

surberj, 2016-02-25:
Structures are slightly blurred. But probably that is normal?

devbharat, 2016-02-25:
I can't really see it at this resolution, but no their shouldn't be significant 'blurring'(You should be able to resolve tiny stuff in undistorted image that you did in the original). Of course, it does some sort of interpolation(you can choose, I've always used linear) between pixels, but that shouldn't blur either.

closed 2016-05-04 by surberj