surberj/Vision #67: GPS measurement

surberj, 2016-05-17:
The current gps module together with the gps_umd ros node produces two issues:

- The uncertainties in x and y are constant over time (variance_x ~ 9m, variance_y ~ 10m). 
- The z measurement is often nan (every second time).

At the moment, I reject all measurements which show a nan value, so the second issue is not too severe, but as I do the keypoint selection based on the uncertainty in x and y the first issue should be solved as soon as possible.

@weblucas did you see these issues as well?

surberj, 2016-05-26:
The gps module on the asctec is clearly better (at least relative positions). 

I am still a bit confused about the units of the uncertainty --> the variance is usually between 0.2 and 2, which is very low if it's unit is meter.

Anyways, we are good with the setup on the asctec for my experiments, so closing here.

weblucas, 2016-05-26:
It is in meters.

closed 2016-12-13 by surberj