surberj/Vision #69: Contribution

surberj, 2016-05-26:
For the final presentation and the potential paper, I should be very clear about my contributions:

First raw ideas:
- With the aid of a reference map and a weak gps prior, we are able to observe the unobservable states (position, heading) of a visual-inertial odometry algorithm.
- We propose an algorithm, which is able to capture the drift in pose estimation, which is an inherent problem of visual-inertial odometry algorithms.
- We demonstrate a new "geometric loop closure" technique by selecting a sub-map based on a gps prior and performing a brute force 2D-3D matching
- We demonstrate the algorithms accuracy to be an order of magnitude higher (?) than gps based localization, enabling autonomous tasks like precise landing
- Scalable to large workspaces

weblucas, 2016-05-26:
in my opinion, to be novel. you need to demonstrate that multiple, non-overlapped geometric loop closures are better than GPS.  Is this public?

surberj, 2016-05-27:
What do you mean exactly with multiple non-overlapping loop closures? 

The accuracy of the estimated pose (OKVIS + GlobalMapAlignment) will depend on the number of successful loop closures. If we get loop closures frequently, the estimated base frame transformation and the final estimated pose are more accurate. If we do not get a loop closure for a minute or two the trajectory drifts away slowly from the ground truth (this will be verified with the experiments with Leica we will do next week). 

If we compare against localization with gps and imu only, we always should be more accurate and moreover, we are able to handle gps losses. If gps is lost, OKVIS still gives reasonable pose estimation over several minutes and also the GlobalMapAlignment is currently set up to look for loop closures (simply within the whole map, instead of a submap). 

This is public, the GlobalMapAlignment node is private at the moment.

closed 2016-12-13 by surberj