surberj/Vision #72: OpenMVG

surberj, 2016-06-06:
Lucas demonstrated on the data of 20160518 flight 0011, that the openMVG library produces pretty nice 3D structure from motion results for the Irchel data:

[image]

The point cloud out of the openMVG pipeline is denser and more accurate in "depth" (world-z) than the OKVIS point cloud (with two drawbacks - more outliers and no scale estimation). The result demonstrates that we should be able to get "better" point clouds out of the data than what I use as reference map at the moment.

Now there is the question how to use the results. I can see different ways:
1) Only for visualization --> Leave the system as it is but use the openMVG map within the rviz visualization
2) Study the openMVG code and modify OKVIS to a "offline, sequentialSfM" version. In ideal case this should leed to a scale consistent, outlier-"free" and more consistent (mainly in depth) reference map.  
3) Modify the openMVG code to describe the points by BRISK features (we need the customized OKVIS 48byte features) and create the reference map with openMVG.

I would best like to try out option 2, as it does not need manual scale estimation but still could give resonable improvement to the reference map.

pmoulon, 2016-06-22:
Hi @surberj 
Just saw that you was doing experiments with OpenMVG to build some MAPS!

What you can do also, is to load the resulting map in your system, extract some features within your library and robustly triangulate them.

Doing 2) is reinventing the wheel, why do you need to spend time to rewrite something that is already working.
Doing 3) Should be fast, you just need to add new region type (for the Brisk feature), then you can use your own feature/descriptor extractor or use the one from opencv.

If you have some scale information, you can simply load the openMVG sfm scene and rescale it.

Would be happy to continue the discussion :+1:

surberj, 2016-06-28:
Hi @pmoulon 
Thank you for your comment! I worked the last days on a presentation, excuse me for my late response. The good thing about the way I created my maps so far (with something similar to https://github.com/ethz-asl/okvis), is, that we include high quality IMU measurements between the image frames into the optimization (we minimize reprojection and IMU error terms). Moreover, there is a meaningful heuristic implemented, which selects frames out of the sequence and marginalizes out frames in between.
The bad thing about it, is, that the code is not optimized for structure-from-motion but rather for real-time localization and mapping, which means that data management gets bad for larger maps (everything is handled within main memory, all tracks are associated with the frames themselves). 

Do you know if there are plans in openMVG to include IMU measurements into the structure-from-motion pipeline? Either directly into the optimization or in a post processing step to align the map with gravity and support scale?

Thanks for your inputs!

pmoulon, 2016-06-28:
Thanks for your answer.
No worries about the delay. One day is better than never ;-)
I plan to add the support of prior (position, rotation) to OpenMVG but after that there is many ways to use those prior.

You can use them as trusted (so they can be directly the global value), or they can be use a relative information in order to make the graph optimization stronger (according user weight confidence).

I would be very happy to make experiments with you if you can share some tiny dataset or point me some dataset we can play with ;-)

Discussion about how handling IMU data as residual cost function would be interesting too since I have not yet experience with it.

We can first work on alignment and scale if you want!
=> If you have two reconstructions with corresponding camera positions (two different coordinates system), the openMVG API is already able to perform the registration.

closed 2016-12-13 by surberj