surberj/Vision #74: Frame is newer than oldest IMU measurement.

maym86, 2016-07-01:
Hi,

I'm getting an error when trying with my own data where the frames aren't processed. It looks like it's to do with the time stamps:

"Frame is newer than oldest IMU measurement. Dropping it."

My cam data 

#timestamp [ns],filename
1467394337801095936,cam0/data/000000000000.jpg
1467394337801126912,cam0/data/000000000001.jpg
1467394337801146880,cam0/data/000000000002.jpg
1467394337801166080,cam0/data/000000000003.jpg
1467394337801186048,cam0/data/000000000004.jpg
1467394337801208064,cam0/data/000000000005.jpg
1467394337801227008,cam0/data/000000000006.jpg
1467394337801244928,cam0/data/000000000007.jpg
1467394337801262080,cam0/data/000000000008.jpg
1467394337801277952,cam0/data/000000000009.jpg
1467394337801295872,cam0/data/000000000010.jpg
1467394337801313024,cam0/data/000000000011.jpg
1467394337801330176,cam0/data/000000000012.jpg
1467394337801346816,cam0/data/000000000013.jpg
1467394337801363968,cam0/data/000000000014.jpg
1467394337801382144,cam0/data/000000000015.jpg
1467394337801406976,cam0/data/000000000016.jpg
1467394337801425152,cam0/data/000000000017.jpg
1467394337801441792,cam0/data/000000000018.jpg
1467394337801457664,cam0/data/000000000019.jpg
1467394337801477120,cam0/data/000000000020.jpg
1467394337801494016,cam0/data/000000000021.jpg
1467394337801511936,cam0/data/000000000022.jpg
1467394337801528832,cam0/data/000000000023.jpg
1467394337801547008,cam0/data/000000000024.jpg
1467394337801563904,cam0/data/000000000025.jpg
1467394337801579776,cam0/data/000000000026.jpg
1467394337801597184,cam0/data/000000000027.jpg
1467394337801614080,cam0/data/000000000028.jpg
1467394337801630976,cam0/data/000000000029.jpg
1467394337801647872,cam0/data/000000000030.jpg
1467394337801666048,cam0/data/000000000031.jpg
1467394337801682944,cam0/data/000000000032.jpg
1467394337801700096,cam0/data/000000000033.jpg
1467394337801716992,cam0/data/000000000034.jpg
1467394337801734912,cam0/data/000000000035.jpg
1467394337801753856,cam0/data/000000000036.jpg
1467394337801771008,cam0/data/000000000037.jpg
1467394337801788160,cam0/data/000000000038.jpg
1467394337801805056,cam0/data/000000000039.jpg
1467394337801822976,cam0/data/000000000040.jpg
1467394337801843968,cam0/data/000000000041.jpg
1467394337801862912,cam0/data/000000000042.jpg
1467394337801882112,cam0/data/000000000043.jpg
1467394337801908992,cam0/data/000000000044.jpg
1467394337801923072,cam0/data/000000000045.jpg
1467394337801942016,cam0/data/000000000046.jpg
1467394337801956864,cam0/data/000000000047.jpg
1467394337801970944,cam0/data/000000000048.jpg
1467394337801985024,cam0/data/000000000049.jpg
1467394337801999104,cam0/data/000000000050.jpg
1467394337802021120,cam0/data/000000000051.jpg
1467394337802035200,cam0/data/000000000052.jpg
1467394337802048768,cam0/data/000000000053.jpg
1467394337802063104,cam0/data/000000000054.jpg
1467394337802077184,cam0/data/000000000055.jpg
1467394337802090752,cam0/data/000000000056.jpg
1467394337802104064,cam0/data/000000000057.jpg
1467394337802117120,cam0/data/000000000058.jpg
1467394337802129920,cam0/data/000000000059.jpg
1467394337802144000,cam0/data/000000000060.jpg
[... 103 lines cut ...]
1467394338172715776,0.0,0.0625,0.125,0.16,-0.33,9.78
1467394338182624000,0.0,0.0,0.0,0.15,-0.32,9.84
1467394338192658944,0.0,0.0,0.0,0.15,-0.34,9.89
1467394338202681088,0.0,0.125,0.0,0.1,-0.36,9.87
1467394338212658944,0.0,0.0,0.125,0.1,-0.35,9.74
1467394338222642944,0.125,0.0625,0.0,0.07,-0.34,9.81
1467394338232637952,0.0,0.0625,0.0,0.1,-0.32,9.83
1467394338242633984,0.0,0.0,0.0,0.09,-0.3,9.86
1467394338252603904,-0.0625,-0.0625,0.0,0.13,-0.31,9.9
1467394338262559744,0.0,-0.1875,0.0,0.13,-0.33,9.72
1467394338272571136,0.0,0.1875,0.0,0.14,-0.35,9.82
1467394338283204096,0.125,0.1875,0.0625,0.09,-0.34,9.87
1467394338292582912,0.0,0.0,-0.0625,0.09,-0.36,9.85
1467394338302606080,-0.0625,-0.125,0.0,0.1,-0.36,9.83
1467394338312642048,0.0625,0.125,0.0,0.13,-0.34,9.73
1467394338322671872,0.0,-0.125,0.0625,0.14,-0.32,9.79
1467394338332698112,0.125,0.0,0.0625,0.12,-0.32,9.83
1467394338342675968,0.0625,0.125,0.0,0.12,-0.32,9.88
1467394338352663040,-0.1875,0.125,0.125,0.15,-0.32,9.87
1467394338362678016,-0.0625,0.0625,0.0,0.13,-0.32,9.73
1467394338372685824,0.0625,0.0,-0.0625,0.12,-0.35,9.86
1467394338382669056,0.0,0.1875,-0.0625,0.11,-0.35,9.83
1467394338392695808,0.0,0.0,0.0,0.1,-0.36,9.89
1467394338402665984,-0.125,-0.0625,0.0,0.13,-0.36,9.89
1467394338412695808,0.0625,-0.0625,0.0,0.13,-0.34,9.76

maym86, 2016-07-01:
Sorry wrong repo

closed 2016-07-01 by maym86

surberj, 2016-07-06:
no problem :)